robotology-legacy/codyco-modules #274: Bug in FT calibration in WholeBodyDynamics

GiulioRomualdi, 2019-03-19:
This morning I tried (with @fjandrad) to calibrate the FT when the robot was in single support. By calling:
```
calibStandingOnOneLink l_sole 300
```
After this, the contact wrench acting on the left foot has a z value that was around zero. Discussing with @gabrielenava  and @fjandrad this should be wrong because in this specific case the force should be equal to 33.3 * 9.81.

Actually, @ahmadgazar (as can be seen in https://github.com/dic-iit/element_jerk-control/issues/25#issuecomment-416877192) was able to perform the calibration on one foot. So something changed in the meantime. 

cc @traversaro @DanielePucci @S-Dafarra

traversaro, 2019-03-19:
> After this, the contact wrench acting on the left foot has a z value that was around zero. Discussing with @gabrielenava and @fjandrad this should be wrong because in this specific case the force should be equal to 33.3 * 9.81.

Which contact wrench?  You read it how?  In which (orientation) frame it is expressed?

traversaro, 2019-03-19:
Which value do you get instead of the expected one?

GiulioRomualdi, 2019-03-19:
Contact wrenches on the left foot expressed in the l_sole frame (the position of the frame and the orientation of the frame are both equals to l_sole)

traversaro, 2019-03-19:
Mhh, can you past the output of wbd (from the logger) during the calibration? Are you sure that all the FT sensors are correctly working?

GiulioRomualdi, 2019-03-19:
>  Are you sure that all the FT sensors are correctly working?

As regards the FT everything seems ok. Indeed if I calibrate the FT when the robot is on the air by using 
```
calib all 300
```
the contact wrenches (both on the left and on the right feet) are fine (i.e. the z forces are more or less 160N)

gabrielenava, 2019-04-03:
We found the origin of the bug. See https://github.com/robotology/yarp/issues/1994

traversaro, 2019-04-03:
As the source of the bug is actually https://github.com/robotology/yarp/issues/1994 , I think we can close this.

closed 2019-04-03 by traversaro