robotology-legacy/codyco-modules #275: wholeBodydynamics configuration files for  iCubLausanne02

MichaelBombile, 2019-05-15:
Hi @traversaro,

We have implemented a whole-body torque control algorithm that work fine in simulation and we would like to test it on the real robot. However, we do not have proper config file (.xml) for the wholeBodyDynamics device to use the robot in torque mode.
By examining the .xml files of iCubDarmstadt01 and iCubGenova02, we created similar files for iCubLausanne02, but we are not sure which .urdf file we should use. 

Thanks in advance.

traversaro, 2019-05-15:
Hi @MichaelBombile, 
 I need to check with the iCub support group about the specific details about iCubLausanne02 (I guess we should be able to add it to https://github.com/robotology/icub-model-generator) however, if I recall correctly iCubLausanne02 is quite a standard v2.5 Ethernet-based iCub without backpack. Just to be safe, you can easily check if the legs are v2.5 if the foot is connected to the leg via a four-bar linkage. 
Based on this, I think you can use either the iCubDarmstadt01 ( https://github.com/robotology/icub-models/tree/master/iCub/robots/iCubDarmstadt01 )  or iCubGenova01 ( https://github.com/robotology/icub-models/tree/master/iCub/robots/iCubGenova01 ) models. The two models should be equal (see https://github.com/robotology/icub-model-generator/blob/bada6e53f0c874d80812a0b3cd146145d1ea85c9/simmechanics/CMakeLists.txt#L265 for the code that generates the  models).

MichaelBombile, 2019-05-15:
Hi @traversaro,

Thanks for your response.
You are right about iCubLausanne02, it is indeed a v2.5 Ethernet-based iCub without backpack. 
It is similar to the iCubDarmstadt01. However, when I tried with iCubDarmstadt01's model, wholeBodyDynamics runs, but as soon as the torso or the legs are sent in torque mode (just using yarpmotorgui), the robot starts shaking (vibrating).  This happens without changing the original PIDTorque gains and with a calibrated robot. 
Did you experience such a situation? or Do you think that it is due to model mismatch?

Cheers,

traversaro, 2019-06-09:
Hi @MichaelBombile , sorry for taking so long to reply. I guess those kind of errors could be related to delays in torque feedback. How are you running the `wholebodydynamics` YARP device?

MichaelBombile, 2019-06-25:
Hi @traversaro, 
Thanks for your reply.
I am running `wholebodydynamics` from an external PC using the command : 
`YARP_ROBOT_NAME=iCubLausanne02 yarprobotinterface --config launch-wholebodydynamics.xml` .
I also think that the delay could cause such a behavior (vibration),

So, would you suggest that I run `wholebodydynamics`  from the robot's head??

Thanks