robotology-legacy/codyco-modules #276: Error when running wholebodydynamics in simulation with iCubGazeboV2_5

gabrielenava, 2019-07-02:
I tried to run a [Yoga](https://github.com/robotology/whole-body-controllers/tree/master/controllers/floating-base-balancing-torque-control) simulation with [iCubGazebov2_5](https://github.com/robotology/icub-models/blob/master/iCub/robots/iCubGazeboV2_5/model.urdf) robot in gazebo. I had an issue when launching the command: `YARP_ROBOT_NAME=iCubgazeboV2_5 yarprobotinterface --config launch-wholebodydynamics.xml` on terminal:

```
[WARNING]Invalid syntax while loading /home/gnava/Software/robotology-superbuild/build/install/share/codyco/robots/iCubGazeboV2_5/launch-wholebodydynamics.xml at line 1 . No DTD found. Assuming version yarprobotinterfaceV3.0 
[ERROR]DTD V1.x has been deprecated. Please update your configuration files to DTD v3.x 
[FATAL]Invalid DTD version, execution stopped.
/home/gnava/Software/robotology-superbuild/build/install/bin/../lib/libYARP_OS.so.3(_ZNK4yarp2os3Log5fatalEPKcz+0x16b) [0x7f0f286c7deb]
yarprobotinterface() [0x431b7d]
yarprobotinterface() [0x41baae]
/home/gnava/Software/robotology-superbuild/build/install/bin/../lib/libYARP_OS.so.3(_ZN4yarp2os8RFModule9runModuleERNS0_14ResourceFinderE+0x62) [0x7f0f28745222]
yarprobotinterface() [0x406195]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f0f27d01830]
yarprobotinterface() [0x406499]

```
Looking at the error message I understood that the follwing line was missing in my `launch-wholebodydynamics.xml` file:

`<!DOCTYPE devices PUBLIC "-//YARP//DTD yarprobotinterface 3.0//EN" "http://www.yarp.it/DTD/yarprobotinterfaceV3.0.dtd">`

Adding the line at the beginning of the file fixed the issue for me, but I don't know how to properly fix the issue upstream. I guess it has to be added for any robot in this repo?

gabrielenava, 2019-07-02:
Another issue: the last device (wholebodydynamics) that should be launched by the `launch-wholebodydynamics.xml` file is actually not launched. Namely, I am talking about this line:  https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95

Here there are no errors: the line is simplyskipped and I have no idea why. But the file exists, and if I substitute the path to the file with the content of the file inside launch-wholebodydynamics.xml` I am finally able to run the Yoga demo. But this is not a good solution.

traversaro, 2019-07-03:
Hi @gabrielenava , 
  please open a different issue for every problem you encounter, thanks! 

For the invalid DTD version, this is due to the change in DTD for yarprobotinterface https://github.com/robotology/yarp/blob/3ce40bff37ba44b1dd1303243b7332a1ce1bd85f/doc/release/v2_3_65.md#yarprobotinterface , and it can be fixed with a commit similar to what @prashanthr05 did in https://github.com/robotology/whole-body-estimators/pull/1/commits/fd3c170d5cd35450439e05bcf5f03eb1015ac438 .

gabrielenava, 2019-07-03:
Separate issue opened here: https://github.com/robotology/codyco-modules/issues/277.

gabrielenava, 2019-07-04:
[PR](https://github.com/robotology/codyco-modules/pull/278) opened.

gabrielenava, 2019-07-12:
PR merged. closing the issue.

closed 2019-07-12 by gabrielenava