robotology-legacy/codyco-modules #277: WholeBodyDynamics device not launched when running launch-wholebodydynamics.xml

gabrielenava, 2019-07-03:
The `wholebodydynamics` device that should be launched by the `launch-wholebodydynamics.xml` file is actually not launched. Namely, I am talking about this line: https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95

Here there are no errors: the line is simply skipped and I have no idea why. But the file exists, and if I substitute the path to the file with the content of the file inside launch-wholebodydynamics.xml` I am finally able to run the Yoga demo. But this is not a good solution.

traversaro, 2019-07-03:
Did you fixed the problem in https://github.com/robotology/codyco-modules/issues/276  ? 
What is location of the files that you are trying to run in the yarprobotinterface?

gabrielenava, 2019-07-03:
I fixed the invalid DTD problem locally on my laptop, see https://github.com/robotology/codyco-modules/issues/276#issue-463294649. I will open a PR for fixing the remote branch too.

the `launch-wholebodydynamics.xml` file is located in `$PATH_TO_SUPERBUILD_INSTALL/share/codyco/robots/iCubGazeboV2_5/` and the `wholebodydynamics-external.xml` is located in `$PATH_TO_SUPERBUILD_INSTALL/share/codyco/robots/iCubGazeboV2_5/estimators`. Superbuild paths seem to be set correctly an I set them by sourcing the [setup.sh](https://github.com/robotology/robotology-superbuild/blob/master/cmake/template/setup.sh.in) file.

traversaro, 2019-07-03:
Have you tried to test it with another  devices file to check if that is loaded instead?

gabrielenava, 2019-07-03:
All the other devices in the file are correctly loaded: https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L89

The `wholebodydynamics` device itself is loaded without any problem if I copy the content of `wholebodydynamics-external.xml` instead of using the path to the xml file as specified currently in the `launch-wholebodydynamics.xml` file: https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95

traversaro, 2019-07-03:
I got it. I asked if you could load any other `devices` files, similar to  `wholebodydynamics-external.xml` but with less devices or something similar.

gabrielenava, 2019-07-03:
I did now the following test:

- I modified the `launch-wholebodydynamics.xml`. I removed [the right foot FT sensor device](https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L89) and I put it in a file called `device-test.xml`. Then, I modified this [path](https://github.com/robotology/codyco-modules/blob/master/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95) to point my `device-test` file instead of `wholebodydynamics-external.xml`. The two files are in the same folder. 

I run `YARP_ROBOT_NAME=iCubgazeboV2_5 yarprobotinterface --config launch-wholebodydynamics.xml`. The last device loaded is the left foot FT:

```
[INFO]created device <analogsensorclient>. See C++ class AnalogSensorClient for documentation.
yarp: Port /wholeBodyDynamics/l_foot_ft_sensor active at tcp://140.93.4.152:10079/
yarp: Port /wholeBodyDynamics/l_foot_ft_sensor/rpc:o active at tcp://140.93.4.152:10080/
yarp: Receiving input from /icubSim/left_foot/analog:o to /wholeBodyDynamics/l_foot_ft_sensor using udp
yarp: Sending output from /wholeBodyDynamics/l_foot_ft_sensor/rpc:o to /icubSim/left_foot/analog:o/rpc:i using tcp
[INFO]created device <analogsensorclient>. See C++ class AnalogSensorClient for documentation.
```

Therefore  it seems the problem is not in the `wholebodydynamics-external.xml` file.

prashanthr05, 2019-07-04:
I also experience this issue.

prashanthr05, 2019-07-04:
I made the following quick hack and it apparently works,

In the `launch-wholebodydynamics-icub-six-fts-sim.xml`
I changed 
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L2
to
``` xml
<robot name="@WBD_YARP_ROBOT_NAME" portprefix="icubSim" build="1"  xmlns:xi="http://www.w3.org/2001/XInclude">
```

Then I changed 
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/launch-wholebodydynamics-icub-six-fts-sim.xml#L95
to
``` xml
    <xi:include href="estimators/wholebodydynamics-external.xml" />
```

In the `wholebodydynamics-icub-external-six-fts-sim.xml`,
I commented the lines ,
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml#L2
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml#L4
and
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml#L75

then finally changed, 
https://github.com/robotology/codyco-modules/blob/75a681590889a13967351911f694dd7a0a9c14df/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-external-six-fts-sim.xml#L5
to
``` xml
<device xmlns:xi="http://www.w3.org/2001/XInclude" name="wholebodydynamics" type="wholebodydynamics">
```

prashanthr05, 2019-07-04:
Were there any recent changes in the parsing done by `yarprobottinterface`?

prashanthr05, 2019-07-04:
In my opinion, the problem lies with the `yarprobotinterface` for the parsing of `<include device /> tags.

GiulioRomualdi, 2019-07-23:
Same problem with the latest version of YARP

GiulioRomualdi, 2019-07-24:
Thanks to @prashanthr05 suggestion the problem is solved. I open a PR

traversaro, 2019-07-29:
Guys, do  you have any idea why this is now necessary? I am a bit out of time to check this out, but it could be interesting.

prashanthr05, 2019-07-29:
> Guys, do you have any idea why this is now necessary? I am a bit out of time to check this out, but it could be interesting.

I faced this problem after updating YARP to 3.2, specifically to this commit https://github.com/robotology/yarp/commit/7ee03f8e042d31dcad271b6782dc1de8398a8b12. and this line

``` xml
<devices file="estimators/wholebodydynamics-external.xml" />
```

in the configuration file was not being parsed by the `yarprobotinterface`.

traversaro, 2019-07-29:
Yes, I was asking because I fear that we are actually doing a workaround but the actual bug is in YARP, but if no one has bandwidth to investigate that I am ok to merge Giulio's  PR. Opening a bug in YARP without a minimun working example is probably just creating noise.

nunoguedelha, 2019-10-21:
I just experienced the same issue. Actually the issue is solved by using the current format supported by YARP for including device config files (we are more familiar with it now).
I just replaced
```xml
<!-- estimators -->
<devices file="estimators/wholebodydynamics-external.xml" />
```
by
```xml
<!-- estimators -->
<devices>
<xi:include href="estimators/wholebodydynamics-external.xml" />
</devices>
```
As it is done in for any other inclusion.
@prashanthr05 had already done this more or less, but I didn't need the other workarounds:
- There is no need for changing the included file for removing the tag `<devices>`.
- The following prefix is no more required
```xml
<device xmlns:xi="http://www.w3.org/2001/XInclude" name=... />
```
- Nor setting the `xmlns:xi` include tool at the top (apparently), probably becasue now it is a default?

I'm pushing a fix in the same PR opened by @GiulioRomualdi https://github.com/robotology/codyco-modules/pull/280.

nunoguedelha, 2019-10-21:
CC @CarlottaSartore

traversaro, 2019-10-22:
Thanks @nunoguedelha

nunoguedelha, 2019-10-22:
Fixed by by #280 .

traversaro, 2019-10-22:
Thanks @nunoguedelha .

closed 2019-10-22 by traversaro