robotology-legacy/codyco-modules #287: Update required iDynTree version

GiulioRomualdi, 2020-01-14:
Since the `iDynTree 1.0.0` has been [released](https://github.com/robotology/idyntree/releases/tag/v1.0.0) 
The required minimum version of iDynTree has been updated. This is required by `COMPATIBILITY SameMajorVersion` 
```cmake 
install_basic_package_files(iDynTree VARS_PREFIX ${VARS_PREFIX}
                                     VERSION ${${VARS_PREFIX}_VERSION}
                                     COMPATIBILITY SameMajorVersion
                                     TARGETS_PROPERTY ${VARS_PREFIX}_TARGETS
                                     NO_CHECK_REQUIRED_COMPONENTS_MACRO
                                     ENABLE_COMPATIBILITY_VARS
                                     DEPENDENCIES ${_IDYNTREE_EXPORTED_DEPENDENCIES})
```
In the main iDynTree [`CMakeLists.txt`](https://github.com/robotology/idyntree/blob/516713a5ec71427ee04746243dbc504ead66862f/CMakeLists.txt#L72-L78) file

Indeed when I try to compile the library I got the following error

```
loSince the `iDynTree 1.0.0` has been [released](https://github.com/robotology/idyntree/releases/tag/v1.0.0) 
The required minimum version of iDynTree has been updated. This is required by `COMPATIBILITY SameMajorVersion` 
```cmake 
install_basic_package_files(iDynTree VARS_PREFIX ${VARS_PREFIX}
                                     VERSION ${${VARS_PREFIX}_VERSION}
                                     COMPATIBILITY SameMajorVersion
                                     TARGETS_PROPERTY ${VARS_PREFIX}_TARGETS
                                     NO_CHECK_REQUIRED_COMPONENTS_MACRO
                                     ENABLE_COMPATIBILITY_VARS
                                     DEPENDENCIES ${_IDYNTREE_EXPORTED_DEPENDENCIES})
```
In the main iDynTree [`CMakeLists.txt`](https://github.com/robotology/idyntree/blob/516713a5ec71427ee04746243dbc504ead66862f/CMakeLists.txt#L72-L78) file

Indeed when I try to compile the library I got the following error

```
loading initial cache file /home/gromualdi/robot-code/robotology-superbuild/build/robotology/WBToolbox/CMakeFiles/YCMTmp/WBToolbox-cache-Release.cmake
-- Found YCM: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YCM (found version "0.11.0.1-20191204+gitad8bb25")
-- Found YARP: /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.3.0+27-20200110.2+git5fbea1e6e")
CMake Error at CMakeLists.txt:35 (find_package):
  Could not find a configuration file for package "iDynTree" that is
  compatible with requested version "0.7.2".

  The following configuration files were considered but not accepted:

    /home/gromualdi/robot-code/robotology-superbuild/build/install/lib/cmake/iDynTree/iDynTreeConfig.cmake, version: 1.0.0
    /home/gromualdi/robot-code/robotology-superbuild/build/robotology/iDynTree/iDynTreeConfig.cmake, version: 1.0.0
```

Related to:
* https://github.com/dic-iit/bipedal-locomotion-controllers/pull/8
* https://github.com/robotology/wb-toolbox/issues/182

@traversaro

traversaro, 2020-01-14:
I think https://github.com/robotology/idyntree/pull/629 is a more sensitive fix, unless 1.0 features are actually needed.

GiulioRomualdi, 2020-01-15:
https://github.com/robotology/idyntree/pull/629 fixed the problem, we can close this issue

closed 2020-01-15 by GiulioRomualdi