robotology-legacy/codyco-modules #288: CMake configuration error - cannot find `paramHelp`

HosameldinMohamed, 2020-01-28:
If I try to compile `codyco-modules` as a standalone repository (outside `robotology-superbuild`). I get the following error during cmake configuration issue:

```
CMake Error at src/modules/torqueBalancing/CMakeLists.txt:10 (find_package):
   By not providing "FindparamHelp.cmake" in CMAKE_MODULE_PATH this project
   has asked CMake to find a package configuration file provided by
   "paramHelp", but CMake did not find one.

   Could not find a package configuration file provided by "paramHelp" with
   any of the following names:

     paramHelpConfig.cmake
     paramhelp-config.cmake

   Add the installation prefix of "paramHelp" to CMAKE_PREFIX_PATH or set
   "paramHelp_DIR" to a directory containing one of the above files.  If
   "paramHelp" provides a separate development package or SDK, be sure it has been installed.
```

`paramHelp` seems to be required to the module `torqueBalancing`.

traversaro, 2020-01-28:
Sorry for the trouble, this is one of the many leftovers. Set the `CODYCO_USES_KDL`  option to `false`.

HosameldinMohamed, 2020-01-28:
> Sorry for the trouble, this is one of the many leftovers. Set the `CODYCO_USES_KDL` option to `false`.

Great! Thank you very much!

I guess we don't need to make a modification since the repo will be migrated. I think we can close this issue.

FYI @prashanthr05

closed 2020-01-28 by HosameldinMohamed