robotology-legacy/codyco-modules #289: [wholeBodyDynamics] add the option to change the type of iDynTree::`UnknownWrenchContact`

HosameldinMohamed, 2020-01-29:
It would be nice if we can assume that only external forces acting on the torso are the thrust forces, for which we know the application point and the force direction, So we can estimate the intensities of more than one wrench per sub-model.

According to the estimation theory behind the algorithm (see the paper [Contact Force Estimations Using Tactile Sensors and Force Torque Sensors](https://www.researchgate.net/profile/Andrea_Del_Prete/publication/236152161_Contact_Force_Estimations_Using_Tactile_Sensors_and_Force_Torque_Sensors/links/00b495166e74a5369d000000.pdf). If the applied external wrenches are assumed to `PURE_FORCE`, the estimator can estimate up to 3 forces per sub-model, and if the applied forces are assumed to be `PURE_FORCE_WITH_KNOWN_DIRECTION`, the estimator can estimate up to 6 forces per sub-model.

The idea is to change the `wholeBodyDynamics` device (Without forcing the software that depends on it to be changed) to have the option to specify the unknown external wrenches from the configuration files.

## Proposed modifications

Here are the proposed modifications to the `wholeBodyDynamics` device:

- The idea is to add some parameters in the configuration file `wholebodydynamics-external.xml` ([example](https://github.com/robotology/codyco-modules/blob/de4546e3f0e4bbf89fc474ccd578febf7836d517/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L9)) right below the [`defaultContactFrames` parameter](https://github.com/robotology/codyco-modules/blob/de4546e3f0e4bbf89fc474ccd578febf7836d517/src/devices/wholeBodyDynamics/app/wholebodydynamics-icub-eth-six-fts.xml#L9).

The additional parameters could be:

- `overrideContactFrames`:
Similar to `defaultContactFrames` contains vector of strings (name of frames). If this parameter exists, the device will ignore the parameter `defaultContactFrames`.

- `contactWrenchType` could be one of the types:
full = FULL_WRENCH
pure = PURE_FORCE
pureKnown = PURE_FORCE_WITH_KNOWN_DIRECTION

- `contactWrenchDirection`:
Which is a vector of 3 real numbers to specify the direction of the contact wrench w.r.t the origin of the corresponding  `overrideContactFrames` frame. They must be filled if the corresponding `contactWrenchType` is `PURE_FORCE`, and be left empty if the type is `FULL_WRENCH` or `PURE_FORCE_WITH_KNOWN_DIRECTION`.

- `contactWrenchPosition`:
Which is a vector of 3 real numbers to specify the position of the contact wrench w.r.t the origin of the corresponding  `overrideContactFrames` frame. This parameter is **optional**. If left empty, the external force will be considered applied to the origin of the link as usual.

## Logic

- If the configuration file doesn't contain the parameter `overrideContactFrames`. It continues, as usual, considering `defaultContactFrames` and assuming `FULL_WRENCH` is applied.

- If the configuration file contains the parameter `overrideContactFrames`. It ignores frames in the parameter `defaultContactFrames` and considers `overrideContactFrames`. It then requires the presence of the next 2 parameters: `contactWrenchType`, and `contactWrenchDirection`

- The device shall check the consistency of data contained in the 3 paramteres: `overrideContactFrames`, `contactWrenchType` and `contactWrenchDirection`.

## Consistency checks

- For each frame in the paramater `overrideContactFrames`, there is one string value in the paramater `contactWrenchType` (one of the 3 types mentioned above), and 3 real numbers in the paramater `contactWrenchDirection`.

- The frames from a single submodel should be no more than 1 if the type is `FULL_WRENCH`, no more than 3 if the type is `PURE_FORCE`, and no more than 6 if the type is `PURE_FORCE_WITH_KNOWN_DIRECTION`.

- Proper storage of the values read from the configuration file.

- Proper filling the object `UnknownWrenchContact`.

HosameldinMohamed, 2020-01-29:
@traversaro @prashanthr05

traversaro, 2020-01-29:
It seems ok!