robotology-legacy/codyco-modules #290: Deprecate codyco-modules

traversaro, 2020-02-01:
As discussed several times, we are finally going to move wholebodydynamics device out of this repo, and move it in https://github.com/robotology/whole-body-estimators . After that, we should be able to finally deprecate the good old codyco-modules (originally born in 2013 as the monolithic `codyco` repo : ) ).

However, I noticed that this repo still installs the two scripts `liftiCubInGui.sh` (a workaround script for visualization in iCubGui) and twoFeetStandingIdleAndCalib.sh, the old "macumba" script to reduce the internal forces on the robot. While  `liftiCubInGui.sh`  can be moved to `whole-body-estimators` as  is meant to be used with `whole-body-dynamics` ,  I do not think that tt make sense to move `twoFeetStandingIdleAndCalib.sh` there.

So, TL;DR question for @robotology/iit-dic : 
* **Are we still using `twoFeetStandingIdleAndCalib.sh` in experiments?** 
* **If the answer to the above is yes, where can we move it? whole-body-controllers?**

traversaro, 2020-02-16:
Given the lack of answers, I assume that the answer is **no** to all answers, and proceed to remove codyco-modules from the superbuild in the next weeks.

traversaro, 2020-02-17:
Probe comment: if you arrived here from the Teams comment, please add a reaction to this issue with 👀 .

S-Dafarra, 2020-02-17:
There is script similar to ``twoFeetStandingIdleAndCalib.sh`` in https://github.com/robotology/whole-body-controllers/blob/master/utilities/releaseLegStressesWhileStanding.sh. The difference is that it just releases the stresses, but it does not calibrate.

traversaro, 2020-03-09:
cc @prashanthr05 @fjandrad

fjandrad, 2020-03-23:
I think we can remove twoFeetStandingIdleAndCalib.sh and keep only the releaseLegStressesWhileStanding.sh

traversaro, 2020-04-05:
Given the discussion and the fact that codyco-modules is going to be removed from robotology-superbuild in https://github.com/robotology/robotology-superbuild/pull/374, I think we can close this issue.

closed 2020-04-05 by traversaro

DanielePucci, 2020-04-05:
So, we got rid of what once upon a time was called "macumba" - and so it was for a long time - with also an [associated soundtrack](https://www.youtube.com/watch?v=7OHZ-s3rtG8), which was really useful during the [CoDyCo project](https://www.youtube.com/watch?v=mGT_G3v6CtA)

CC
@francesco-romano @jeljaik @iron76